TimSoethout/goodwe-sems-home-assistant #88: Error while setting up sems platform for sensor

reindruu, 2023-09-27:
Hi,
I'm using the GoodWe SEMS PV API for more than a year now.
Suddenly last night the API stopped working.
I tried the usual things such as reboot, restart, reload etc.
The error in the Home Assistant Core is:

> Error while setting up sems platform for sensor
> Traceback (most recent call last):
>   File "/config/custom_components/sems/sensor.py", line 57, in async_update_data
>     inverters = result["inverter"]
>                 ~~~~~~^^^^^^^^^^^^
> TypeError: 'NoneType' object is not subscriptable
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 293, in _async_refresh
>     self.data = await self._async_update_data()
>                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 248, in _async_update_data
>     return await self.update_method()
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/config/custom_components/sems/sensor.py", line 97, in async_update_data
>     raise UpdateFailed(f"Error communicating with API: {err}")
> homeassistant.helpers.update_coordinator.UpdateFailed: Error communicating with API: 'NoneType' object is not subscriptable
> 
> The above exception was the direct cause of the following exception:
> 
> Traceback (most recent call last):
>   File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 359, in _async_setup_platform
>     await asyncio.shield(task)
>   File "/config/custom_components/sems/sensor.py", line 118, in async_setup_entry
>     await coordinator.async_config_entry_first_refresh()
>   File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 264, in async_config_entry_first_refresh
>     raise ex
> homeassistant.exceptions.ConfigEntryNotReady: Error communicating with API: 'NoneType' object is not subscriptable
> 

Can someone help?

reindruu, 2023-09-27:
And 15 hours later it works again.
During those 15 hours the SEMS portal site was reachable normal. On my network and also outside.
I think GoodWe had some maintenance again. They have maintenance a lot, but then the SEMS portal has also problems.
If it happens again I will ask GoodWe support. I do not expect anything, but who knows......
I will close this tomorrow, maybe someone wants to add something.

cthulu, 2023-09-27:
Yeah it works for me as well now. Strange

reindruu, 2023-09-29:
Closed because suddenly solved....

closed 2023-09-29 by reindruu