TimSoethout/goodwe-sems-home-assistant #89: Feedback on moving from extra_state_attributes to sensors.

L-four, 2023-10-21:
I have been working on a branch which exposes API data as Sensors rather then extra_state_attributes on the inverter entity. 
Is allows us to tell home assistant about the types of the attributes.

Here's where I currently stand in this development https://github.com/L-four/goodwe-sems-home-assistant/blob/master/custom_components/sems/sensor.py

I would like to gauge interest merging this change and gathering feedback on these changes.

Screenshots:

[image]
[image]

TimSoethout, 2023-10-23:
Really nice work. I think it would definitely make the component more future proof! Also it makes access to the different attributes as sensors a lot nicer than using templates.

Maybe a next step would be to make a PR and iterate in there, or put it on a branch here, so we can make a beta release at least.

f00f, 2023-10-31:
Yeah, individual sensors would be awesome.
I'm a noob with HA, but I could help with testing this anyways - if someone tells me how to install a dev branch of this integration.

robinhood-code, 2023-11-03:
this is nice. hope it will be merged soon.

closed 2026-02-06 by TimSoethout