TimSoethout/goodwe-sems-home-assistant #94: Sems Portal

davidsonimagerygmailcom, 2023-12-31:


davidsonimagerygmailcom, 2023-12-31:
So sorry for the edits, git hub "post" button wasn't functonal (known issue with github) and I was trying to get post to work with less text (which worked).
Screenies ;
[image]

[image]

[image]
[image]

skreza, 2024-01-02:
I'm also seeing this exact same issue since 28th Dec.
Agree with @davidsonimagerygmailcom that it's likely an issue on the SEMS portal side, maybe they need to configure NTP on their servers?

Unfortunately I can't see a way to use an automation to disable the SEMS service for a period of time, as that would have been a quick workaround within HA. Only other thing I can thing of as a workaround for now is to create a new template sensor in HA that copies the value of the original sensor from this integration and stop this sensor from updating between 00:00 - 00:15, you would then have to add this sensor to the energy dashboard for it to reflect correctly. Hardly ideal but would achieve the desired outcome!

Any help from @TimSoethout would be greatly appreciated!

davidsonimagerygmailcom, 2024-01-02:
As a workaround I've done a few things, as yet I am not 100% sure if it's sorted it (and its only a bandaid).

So the issue as ew know is the double dip ... the import kwh and export kwh all go to zero around midnight, then dip back to the preivous days usages before again going to zero (all of which is data coming from Sems Portal, yet doens't show up in the reports you can run from it).    SemsPortal accounting bug of some sort which came in as soon as they fixed the bigger bug which was KWH metrics weren't resetting at all (for days, even on semsportal).

So the issue is HA seeing that zero'ing, followed by data - and thinks its new usage (hooray for generating power from the moon right??).

I did initially try using an automation that does a Utlility meter reset (as pictured below) ... but even so the "consumed" power would still show (and util meter calibrate doesnt apply to that).

So I went a step further and actually turned off the recorder from 11.59pm to 1230am, my idea being not recording any energy data at all therefore it disregards what it sees.

If this doesnt work, another idea I had was I might just disallow HA from being able to talk to the internet between midniht ad 1230 ... so it never gets exposed to this garbage double dip (it's back to zero by 1215 usually).

I reckon Goodwe have a bug in their system which started around Christmas Eve ... they've "fixed" it by zero'ing values a second time instead of fixing the original fault (maybe unwittingly), and it doesn't affect the data they present to users so probably don' care too much for what HA is pulling from them via API. 

**I've not yet confirmed if this actually works (before anyone goes following down this line).**

[image]
[image]
[image]

[image]

skreza, 2024-01-02:
Thanks for the update @davidsonimagerygmailcom

I've come up with a similar workaround for now and will update tomorrow if it has helped.
Essentially I've disabled update polling on the integration itself and setup an automation to update the entities every minute between 00:15:00 through to 00:00:30 which hopefully gets the 0kWh at midnight but skips everything between 12:01AM to 12:15AM

YAML for the automation below, you must have disabled polling on the integration first (Open integration, click 3 dot menu, system options, turn off 'Enable polling for updates') Also untested so don't do this if you're unsure, I'll provide an update tomorrow advising if I had any luck with this or not.

Automation is triggered every minute to update the entities in the list. You'll need to change these to match your entity ids, I just added all entities associated with the integration.

`alias: GoodWe SEMS Portal Polling
description: ""
trigger:
  - platform: time_pattern
    seconds: "00"
condition:
  - condition: time
    after: "00:15:00"
    before: "00:00:30"
action:
  - service: homeassistant.update_entity
    data: {}
    target:
      entity_id:
        - sensor.homekit_xyz
        - sensor.homekit_xyz_export
        - sensor.homekit_xyz_import
        - sensor.inverter_xyz
        - sensor.inverter_xyz_energy
mode: single`

davidsonimagerygmailcom, 2024-01-02:
You could even apply the logic of missing the first zero'ing entirely (as you'll get it after 1215 anyway) ... kinda why I went for 1159pm (bypass the whole shemozzle)

skreza, 2024-01-02:
I wasn't sure if it needed to reset to zero at midnight for the energy dashboard to calculate it correctly.
Assuming it doesn't then absolutely the condition could be changed to 12:15am through to 11:59pm
```
condition:
  - condition: time
    after: "00:15:00"
    before: "23:59:00"
```

davidsonimagerygmailcom, 2024-01-02:
Just to add spanner to the works, I notice the first zero'ing was about 00:01:22 ... then 00:02:24 back up to 10.5kwh ... 00:06:02 back to 0kwh

The kwh readings aren't the same either so not sure what's going on back end.

[image]

davidsonimagerygmailcom, 2024-01-02:
> I wasn't sure if it needed to reset to zero at midnight for the energy dashboard to calculate it correctly. Assuming it doesn't, then absolutely the condition could be changed to 12:15am through to 11:59pm
> 
> ```
> condition:
>   - condition: time
>     after: "00:15:00"
>     before: "23:59:00"
> ```

Good point.   Not sure!

davidsonimagerygmailcom, 2024-01-02:
> alias: GoodWe SEMS Portal Polling
> description: ""
> trigger:
>   - platform: time_pattern
>     seconds: "00"
> condition:
>   - condition: time
>     after: "00:15:00"
>     before: "00:00:30"
> action:
>   - [...]

Where / how do you define your YAMl code? is it on the Automation itself, or into a file and then calling it somehow?
[image]

skreza, 2024-01-02:
Create a new automation as normal, then press three dots in top right and select Edit in YAML.
Paste the code replacing anything already visible and then change it back to Edit in visual editor from that top menu again.

davidsonimagerygmailcom, 2024-01-02:
How would we define every 30 seconds for polling instead of minute? 

I wonder if ;

trigger:
  - platform: time_pattern
    seconds: "00"
    seconds: "30"

is permissible

(please excuse my amateurism!)

skreza, 2024-01-02:
Ok, so based on the fact that we don't really know when the data will spike after zeroing the first time and that the first zeroing may not be exactly at midnight I've modified things slightly and added a second automation. The first automation is same as above however I've removed the condition so it'll run every minute no matter what. The second automation will trigger when the entity hits 0, it will then disable the first automation wait fifteen minutes and then enable it again. This way we don't have to worry about exact timing from the SEMS side.

**Automation 1**
```
alias: GoodWe SEMS Portal Polling
description: ""
trigger:
  - platform: time_pattern
    seconds: "00"
condition: []
action:
  - service: homeassistant.update_entity
    data: {}
    target:
      entity_id:
        - sensor.homekit_xyz
        - sensor.homekit_xyz_export
        - sensor.homekit_xyz_import
        - sensor.inverter_xyz
        - sensor.inverter_xyz_energy
mode: single
```

**Automation 2**
```
alias: GoodWe SEMS Portal Polling Controller
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.homekit_xyz_export
    to: "0"
condition: []
action:
  - service: automation.turn_off
    target:
      entity_id: automation.goodwe_sems_portal_polling
    data:
      stop_actions: true
  - delay:
      hours: 0
      minutes: 15
      seconds: 0
      milliseconds: 0
  - service: automation.turn_on
    target:
      entity_id: automation.goodwe_sems_portal_polling
    data: {}
mode: single
```

skreza, 2024-01-02:
Re 30 second polling, you would just need a second action, one covers 00 seconds the other covers 30 seconds.
```
trigger:
  - platform: time_pattern
    seconds: "00"
  - platform: time_pattern
    seconds: "30"
```

davidsonimagerygmailcom, 2024-01-02:
Done and working well, i might just to go straight to using your method and see what happens tonight (it's 3pm here).

micheldejongh, 2024-01-02:
Another option: 

trigger:
  - platform: time_pattern
    seconds: /30

davidsonimagerygmailcom, 2024-01-02:
Skreza's workaround worked perfectly

Stopped polling a bit before midnight... oblivious to the double dipping being done at Goodwe's Cloud , and then at 1220 polls again to observe a zero and move on

[image]

skreza, 2024-01-03:
Glad to hear it worked @davidsonimagerygmailcom Now I need to figure out why it didn't work for me! Hahaha.
Did you end up making it time based entirely or did you use the last method I posted which looked for 0 kWh and then disabled the polling automation for 15 minutes?
I tried this way and it did hit 0 however it didn't seem to trigger that automation for some reason, last triggered should be around 12 hours ago not 19! Hmmmm
[image]

davidsonimagerygmailcom, 2024-01-03:
Hey Mate, I just went with your first solution ... the simple one.  
It just polls all day long (every 30 seconds), except betweeen 23:59 and 00:20 ... worked a treat.

[image]
[image]
[image]
[image]

skreza, 2024-01-03:
Thanks for that, I might have made it more complex than it needed to be haha!
Reverted it back to time based with the condition, hopefully all good tonight!

davidsonimagerygmailcom, 2024-01-03:
Fingers Crossed!
All crusing along on my side.  I guess ew won't actually know if the double dip gets fixed until we go back to polling during 2359 > 0020hrs

[image]

justinmaiuto, 2024-01-03:
Just found this one, seems similar issue (possibly the same?)

https://github.com/home-assistant/core/issues/106677

TimSoethout, 2024-01-03:
Hi all, great that you've found a workaround. I'm currently limited in time and energy to work on this, but I'm happy to merge a fix if someone provides it. :)

davidsonimagerygmailcom, 2024-01-03:
> Just found this one, seems similar issue (possibly the same?)
> 
> [home-assistant/core#106677](https://github.com/home-assistant/core/issues/106677)

Yeeepp that's what first caught my eye ... I was perplexed how I'd made so much power from the Moon, and who was running around pulling huge loads at midnight :)

skreza, 2024-01-03:
Ok went back to my original suggestion and worked perfectly last night.

So for those looking for a workaround, the YAML code below works well, create a new automation and use this, make sure you change the sensors in the entity_id list to match your sensors. You also need to disable polling on the 'GoodWe SEMS PV API' integration (Open integration, click 3 dot menu, system options, turn off 'Enable polling for updates')

```
alias: GoodWe SEMS Portal Polling
description: ""
trigger:
  - platform: time_pattern
    seconds: "00"
condition:
  - condition: time
    after: "00:20:00"
    before: "23:59:00"
action:
  - service: homeassistant.update_entity
    data: {}
    target:
      entity_id:
        - sensor.homekit_xyz
        - sensor.homekit_xyz_export
        - sensor.homekit_xyz_import
        - sensor.inverter_xyz
        - sensor.inverter_xyz_energy
mode: single
```

chicaneau, 2024-01-10:
Thanks for the automation to resolve the midnight double dip issue @skreza. I have also noticed that at around the same time this issue started, i am now seeing a gap in my house consumption between 10pm and midnight, is anyone else seeing this too? it's happening on my main SEMS homekit integration sensor sensor.homekit_91000hkuxxxxxxx. Any thoughts? 

It was happening prior to disabling to my polling and moving to the automation for updating entity. I should note that my grid import stats are unaffected between 10pm and midnight which makes this ever stranger.
[image]
House Consumption

[image]
Grid Import

justinmaiuto, 2024-01-10:
> Thanks for the automation to resolve the midnight double dip issue @skreza. I have also noticed that at around the same time this issue started, i am now seeing a gap in my house consumption between 10pm and midnight, is anyone else seeing this too? it's happening on my main SEMS homekit integration sensor sensor.homekit_91000hkuxxxxxxx. Any thoughts?
> 
> It was happening prior to disabling to my polling and moving to the automation for updating entity. I should note that my grid import stats are unaffected between 10pm and midnight which makes this ever stranger. ![house consumption](https://private-user-images.githubusercontent.com/26127713/295402700-e5588119-3870-4750-947c-4c47a567153a.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ4NTM1NTMsIm5iZiI6MTcwNDg1MzI1MywicGF0aCI6Ii8yNjEyNzcxMy8yOTU0MDI3MDAtZTU1ODgxMTktMzg3MC00NzUwLTk0N2MtNGM0N2E1NjcxNTNhLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEwVDAyMjA1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ3NzI2ZjZjOTBmMjY4NTZjMzIxNjU3YTIwODU3NzY3ZWNkN2VkZThlN2I3MWM3NjUxNWUwNGY4ZGU5OWY0YmUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.WcrYc5opytvM3VgDF1rgoih57PvsXCJzOG_VR-XVWEQ) House Consumption
> 
> ![grid import](https://private-user-images.githubusercontent.com/26127713/295403804-524a2ee9-707c-4f60-bbe8-ed9a0a46f52c.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ4NTM1NTMsIm5iZiI6MTcwNDg1MzI1MywicGF0aCI6Ii8yNjEyNzcxMy8yOTU0MDM4MDQtNTI0YTJlZTktNzA3Yy00ZjYwLWJiZTgtZWQ5YTBhNDZmNTJjLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEwVDAyMjA1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc4MGQ3NTQ4ZGY2NTZjMWZkNjdhOWVlMmYxZDEyMDViZjZkN2VkOWM1MDUxMGIwMWQ2Nzg1ZWYxNzg5MDk1ZTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.E-rHosMO4TavO-5gpPtwBiUlDnjH4EPdUDHIbPNGgD0) Grid Import

Im also seeing this, but it is for the 'current consumption' value, not the running daily total which the energy dashboard uses. So its not impacting the $ calc on the enegy dashboard, but is annoying that the other dashboard I setup has a gap in data

skreza, 2024-01-10:
Confirming same issue for me also, however my data is blank from 9pm - midnight, I wonder if this is time zone based? I'm in Melb so currently +11 UTC.

I hadn't noticed this as I don't look at house consumption all that much, I have another integration that pulls data from a Powerpal on my meter and tend to look more at grid import power rather than ongoing house consumption. As @justinmaiuto mentioned, the total consuming is still correct though so data in energy dashboard should still be pretty accurate.

Data in SEMS portal looks fine, pretty weird issues but unfortunately I'm no dev, so trying to fix the source code is above my expertise!

chicaneau, 2024-01-10:
> Confirming same issue for me also, however my data is blank from 9pm - midnight, I wonder if this is time zone based? I'm in Melb so currently +11 UTC.
>
> I hadn't noticed this as I don't look at [...]

so you lose 3 hours of data? weird! Im brisbane so im +10 UTC

davidsonimagerygmailcom, 2024-01-10:
No issue here on that 9pm to midnight bit
[image]

The double dip still occurs though (for humour last night I turned the polling back on between 2359 > 0015, double dipping still there).

davidsonimagerygmailcom, 2024-01-15:
Right I'm seeing weird things were house load is reading the same value for hours on end.  THought it was HA, but sems portal shows the same thing.  Yet, the graph is in sems portal shows the correct house load changing.  Am I missing something here? Only seemed to break down today.   Not a polling issue because the consumtion counters are increasing (and in HA too)
[image]

justinmaiuto, 2024-01-17:
> Right I'm seeing weird things were house load is reading the same value for hours on end. THought it was HA, but sems portal shows the same thing. Yet, the graph is in sems portal shows the correct house load changing. Am I missing something here? Only seemed to break down today. Not a polling issue because the consumtion counters are increasing (and in HA too) ![2024-01-15_12-04-05](https://private-user-images.githubusercontent.com/116411962/296651249-1eb368d6-1b26-4ae8-8937-ece5ea2309cf.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU0NTAzNzQsIm5iZiI6MTcwNTQ1MDA3NCwicGF0aCI6Ii8xMTY0MTE5NjIvMjk2NjUxMjQ5LTFlYjM2OGQ2LTFiMjYtNGFlOC04OTM3LWVjZTVlYTIzMDljZi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExN1QwMDA3NTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lNWEwOTk0ZTcxY2Y0YmY1NTM2NDg5YmQ1Y2Y4MjBjNWUxMTE0ZmJkMzYzNjhjMjgwNDM1YTA0NDZiYzYwZDQxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.qeWiHYYvOLQ9Lqf34_hqcjHZin3SWlKfn_uyZov8tn0)

Yep, I noticed the same in SEMS portal. The graph is ok, but the text value gets stuck. This coincides with what happens in HA.

davidsonimagerygmailcom, 2024-01-20:
The double dip issue is still there, tested last night.

Goodwe support (when they choose to respond to repeated followups) simply say "I can see from my end that the inverter is back online and the readings are showing.
And regarding the double dip issue is it still presisting ? because from my end i am seeing that it doesn't do that and that it restarts to zero at midnight with no problem"

Without any regard to the extensive outlining of my information prior to them.  The issue doesn't appear to SemsPortal because whatever this "zero'ing" procedure is that they're doing, it's not being ingested by the sems graph / portal data.   

Have repeatedly asked to have the issue escalated to the team whom actually look after the back end databases but falls on deaf ears.

back to leaving the polling off between 2359 and 0020

davidsonimagerygmailcom, 2024-01-23:
Right, they've shifted some of the dysfunction to earlier ... now seeing a big change up at around 11.26pm

Unsure if there's still further dips after midnihgt because I've still got polling turned off.

Sounds like someone at goodwe is really struggling with the data roll routines

Will alter my workaround to stop polling between say 2320 and 0020 (though am tempted to poll constantly and get a fresh picture of what they'er doing).

[image]

davidsonimagerygmailcom, 2024-01-23:
Deleted the last comment about 1130pm uptick, as realised internet had been down and homekit wasn't reporting usage for several hours.   Disregard for anyone that read it before I deleted.

BrutalDeluxe20, 2024-02-01:
I was wondering about the update interval for the automation @skreza posted.  Looking at SEMS portal the data seems to be written every 5 minutes, so would it make sense to have the automation run every 5 mins to match that?  I figure if I can avoid running it more than it needs that can only be a good thing?

BrutalDeluxe20, 2024-02-02:
Thanks everyone for your input into this.  @skreza I've implemented your automation and data looks spot on today, so thanks for sharing.

BrutalDeluxe20, 2024-10-29:
Hi @TimSoethout, I seem to be having issues with this after daylight savings has started here in Melbourne Australia.  I've changed the integration to stop upating the entities now from 1am - 1:10am (ie. what was previous 12am - 12:10am before daylight savings), but it doesn't seem to be working?
Any ideas on what is going on?  Unless whereever I'm pulling the data from has started daylight savings at the same time therefor the time difference is further skewed? Thanks a lot

Before daylight savings started
[image]

Today
[image]

davidsonimagerygmailcom, 2024-10-29:
Hmm, no DLS here in WA, but I'm seeing similiar weird things with big spikes 3am, then nothing either side.
Might not be the original issue this thread refers to, but usual SemsPortal antics?
It really shouldn't be this difficult but whoever is handling the mathematical / database rolling and management really seem to struggle. 
[image]

skreza, 2024-10-29:
@BrutalDeluxe20 I'm Melb based and haven't had any issues post daylight savings.

Based on previous comments, you mentioned you changed to automation polling rather than using the SEMS polling in the integration. How have you got it configured currently? Mine is still on the automation as per: https://github.com/TimSoethout/goodwe-sems-home-assistant/issues/94#issuecomment-1876042918

BrutalDeluxe20, 2024-10-29:
Hi @skreza, yes I have it setup as you do.  So have disabled polling in the GoodWe SEMS API integration and then use an automation to run every minute to manually update the various entities with the break around midnight.  I did only have 10 minutes I think, so have extended that to 20 as per comment #94 and will see how it goes tonight.  Just strange it seemed to stop working for me when DLS started?

TimSoethout, 2024-10-29:
> Hi @TimSoethout, I seem to be having issues with this after daylight savings has started here in Melbourne Australia. I've changed the integration to stop upating the entities now from 1am - 1:10am (ie. what was previous 12am - 12:10am before daylight savings), but it doesn't seem to be working? Any ideas on what is going on? Unless whereever I'm pulling the data from has started daylight savings at the same time therefor the time difference is further skewed? Thanks a lot
> 
> Before daylight savings started ![Previous](https://private-user-images.githubusercontent.com/147672963/380980334-c25daa71-35d3-488c-baca-e4effcb1a6e9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAyMDY2MDIsIm5iZiI6MTczMDIwNjMwMiwicGF0aCI6Ii8xNDc2NzI5NjMvMzgwOTgwMzM0LWMyNWRhYTcxLTM1ZDMtNDg4Yy1iYWNhLWU0ZWZmY2IxYTZlOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAyOVQxMjUxNDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZWE2MTQ1ODYxZjA3OGVjNmFkN2FmYjZlOGRhMzczNjk1NzM0YmJkYWYwMzBkYWEwZDc1ZDRmN2M2ZGFiMWVmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.w0mdp-48criFnCdkAk09KAXm5hQGQHciImxfO_sjNI8)
> 
> Today ![Today](https://private-user-images.githubusercontent.com/147672963/380980425-152c3ac7-e812-4637-a24b-a77d622b885e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAyMDY2MDIsIm5iZiI6MTczMDIwNjMwMiwicGF0aCI6Ii8xNDc2NzI5NjMvMzgwOTgwNDI1LTE1MmMzYWM3LWU4MTItNDYzNy1hMjRiLWE3N2Q2MjJiODg1ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAyOVQxMjUxNDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04YWMxZmE2ZDU2ZDQyOGI4NmM2OWU3OTdmN2YxZjgyMjUwYTQyNDRlYjllNzQ2ODYyOWFlYzUzOTMxMGRhMzFmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.JMqE_n7WUGXYzWY85NhpALKS6sy5j9mBBOxG8JSYwn4)

It also could be ralated to the component only recovering after some time, and counting all the previous hours to the current hour. Sometimes you get graphs like that then.

BrutalDeluxe20, 2026-02-07:
@TimSoethout should we be updating to 8.2.0 and disable the 1 minute update automation and email polling again?

I'm still on 7.4.1 and have left the 1 minute update automation running and kept polling disabled. Thanks

BrutalDeluxe20, 2026-02-07:
I've updated to 8.2.0 and have enabled polling and disabled the 1 min automation and will report back on how it behaves tonight

RobSmyth, 2026-02-07:
> [@TimSoethout](https://github.com/TimSoethout) ..... I'm using HomeKit and I noticed my energy dashboard didn't update properly in the hour after upgrading. I've tried removing it all and re-installing but didn't fix it

I have found it often takes hours before it starts working (HomeKit) again. For me restarting etc does not help. Feels like the SEMS API takes a long time to release a token sometimes.

RobSmyth, 2026-02-07:
Looking at the changelog I see that 8.2.0 has breaking changes. e.g: `The inverter control switch device class changed from OUTLET to SWITCH; If you have automations/dashboards that filter by device class outlet, update them to use device class switch`

I forget, does this project use Semantic Versioning?

BrutalDeluxe20, 2026-02-07:
> I have found it often takes hours before it starts working (HomeKit) again. For me restarting etc does not help. Feels like the SEMS API takes a long time to release a token sometimes.

Ok thanks, I'll leave it overnight and check in the morning

GertJanH, 2026-09-27:
Still happening after the SEMS+ migration, now on the SEMS+ Web counters (11.11.0, EU, single GW3000-XS). On 2026-09-27 at 00:01 CEST, one poll returned the previous day's production added once more to the week/month/year/lifetime counters, and the next poll (00:02) was back to normal:

- `etotal`: `1517.2 → 1528.2 → 1517.2`
- `energy_this_month`: `245 → 256 → 245`
- `energy_this_week`: `62.7 → 73.7 → 62.7`

The +11.0 is exactly the production of 2026-09-26. For `etotal` and `month`, the dip is under 10%, so Home Assistant ignores it. For `energy_this_week`, the drop is large enough to be booked as a meter reset, adding 73.7 kWh to its statistics. The same happened to `this_week` on 2026-09-25 (+49.4 kWh) and to `this_month` on 2026-09-23 (+6.2 kWh).

TimSoethout, 2026-09-27:
I have published a fix in the [11.12.0-beta.2 pre-release](https://github.com/TimSoethout/goodwe-sems-home-assistant/releases/tag/11.12.0-beta.2).

The SEMS+ counter handling now holds an increase for one poll before publishing it. If the next poll returns to the previous value, the transient spike is discarded; if the increase persists, it is accepted. This should prevent the reported 00:01 spike from being treated by Home Assistant as a meter reset. Confirmed period changes are still allowed to reset the daily, weekly, monthly, and yearly counters.

If you can test this beta, please let me know whether the next midnight rollover avoids the extra Recorder statistics. HACS users need to enable pre-releases to install it.

RobSmyth, 2026-09-27:
> I have published a fix in the [11.12.0-beta.2 pre-release](https://github.com/TimSoethout/goodwe-sems-home-assistant/releases/tag/11.12.0-beta.2).
>
> The SEMS+ counter handling now holds an increase [...]

If SEMS+ has the same SEMS issue as we observed I do not think this solution would work as.

1. Past experience is that the incorrect values are reported multiple times and can it for a few minutes (multiple polls). e.g: [comment-1873654531](https://github.com/TimSoethout/goodwe-sems-home-assistant/issues/94#issuecomment-1873654531)

2. The poll rate is around 30/60 seconds so, that approach would not be good for me as it would introduce a significant delay and I'm already finding the 1-2 minute delay very frustration. An extra delay is not acceptable for me.

Suspending polling, as I do, from 23:58 to 00:20 has proven 100% good over a few years now. No solar at this time, i've never been looking at HASS over this time.

```
id: '1713086828488'
  alias: GoodWe SEMS Portal Polling
  description: Poll SEMS
  triggers:
  - seconds: '30'
    trigger: time_pattern
  conditions:
  - condition: time
    after: 00:20:00
    before: '23:58:00'
    weekday:
    - sun
    - mon
    - tue
    - wed
    - thu
    - fri
    - sat
  actions:
  - target:
      entity_id:
      - sensor.solar_inverter_energy
      - sensor.smart_meter
    data: {}
    action: homeassistant.update_entity
  mode: single
```

GertJanH, 2026-10-01:
Confirming the fix on **11.12.0-beta.4 → beta.7** (GW3000-XS-30). Three nights in a row clean since 29-09, including tonight's combined day + month reset: `today` and `this_month` went to 0 at 00:20, `week`/`total` kept their values, no double counting and no statistics corrections needed. On 11.11.0 this went wrong four times in one week. HA even happened to restart at 00:00:51, inside the hold window: the counters stayed unavailable until 00:20 and then came back correct.