TimSoethout/goodwe-sems-home-assistant #96: depreciated constants HA Core 2025.1

HansOtten, 2024-01-06:
Logger: homeassistant.const
Source: helpers/deprecation.py:205
First occurred: 13:48:09 (4 occurrences)
Last logged: 13:48:09

DEVICE_CLASS_POWER was used from sems, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorDeviceClass.POWER instead, please create a bug report at https://github.com/TimSoethout/goodwe-sems-home-assistant/issues
POWER_WATT was used from sems, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfPower.WATT instead, please create a bug report at https://github.com/TimSoethout/goodwe-sems-home-assistant/issues
DEVICE_CLASS_ENERGY was used from sems, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorDeviceClass.ENERGY instead, please create a bug report at https://github.com/TimSoethout/goodwe-sems-home-assistant/issues
ENERGY_KILO_WATT_HOUR was used from sems, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfEnergy.KILO_WATT_HOUR instead, please create a bug report at https://github.com/TimSoethout/goodwe-sems-home-assistant/issues

HansOtten, 2024-01-06:
Logger: homeassistant.components.sensor
Source: helpers/deprecation.py:205
Integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 13:48:09 (2 occurrences)
Last logged: 13:48:09

STATE_CLASS_TOTAL_INCREASING was used from sems, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorStateClass.TOTAL_INCREASING instead, please create a bug report at https://github.com/TimSoethout/goodwe-sems-home-assistant/issues

berkavil, 2024-01-18:
Hallo, +1 here,
I started to receive the same Warning, concerning more variables/constants as is listed below:

Logger: homeassistant.const
Source: helpers/deprecation.py:205
First occurred: 12:17:18 (10 occurrences)
Last logged: 12:17:20

POWER_WATT was used from sems, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfPower.WATT instead, please create a bug report at https://github.com/TimSoethout/goodwe-sems-home-assistant/issues
DEVICE_CLASS_ENERGY was used from sems, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorDeviceClass.ENERGY instead, please create a bug report at https://github.com/TimSoethout/goodwe-sems-home-assistant/issues
ENERGY_KILO_WATT_HOUR was used from sems, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfEnergy.KILO_WATT_HOUR instead, please create a bug report at https://github.com/TimSoethout/goodwe-sems-home-assistant/issues
TEMP_CELSIUS was used from spaclient, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfTemperature.CELSIUS instead, please create a bug report at https://github.com/plmilord/Hass.io-custom-component-spaclient/issues
TEMP_FAHRENHEIT was used from spaclient, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfTemperature.FAHRENHEIT instead, please create a bug report at https://github.com/plmilord/Hass.io-custom-component-spaclient/issues

I hope we will be able still to use this integration after Jan-2025
thx.
Willy

lindsand, 2024-02-06:
As a hint for code changes:

The sensors are defined in sensor.py

https://github.com/TimSoethout/goodwe-sems-home-assistant/blob/master/custom_components/sems/sensor.py

from homeassistant.components.sensor import STATE_CLASS_TOTAL_INCREASING, SensorEntity
from homeassistant.const import (
    DEVICE_CLASS_POWER,
    POWER_WATT,
    CONF_SCAN_INTERVAL,
    DEVICE_CLASS_ENERGY,
    ENERGY_KILO_WATT_HOUR,

and valid values are listed here (where warnings for version 2025.1 are mentioned):

https://github.com/home-assistant/core/blob/dev/homeassistant/const.py

...

Power units
class UnitOfPower(StrEnum):
"""Power units."""

WATT = "W"
KILO_WATT = "kW"
BTU_PER_HOUR = "BTU/h"
_DEPRECATED_POWER_WATT: Final = DeprecatedConstantEnum(
UnitOfPower.WATT,
"2025.1",
)
"""Deprecated: please use UnitOfPower.WATT."""
_DEPRECATED_POWER_KILO_WATT: Final = DeprecatedConstantEnum(
UnitOfPower.KILO_WATT,
"2025.1",
)

TimSoethout, 2024-03-11:
Fixed by  #102

closed 2024-03-11 by TimSoethout