TimSoethout/goodwe-sems-home-assistant #99: HomeKit withtout sn

dev-null-undefined, 2024-01-23:
jq ".data.homKit" data.json
```json
{
  "homeKitLimit": false,
  "sn": null
}
```
jq ".data.powerflow" data.json  
```json
{
  "pv": "381.9(W)",
  "pvStatus": -1,
  "bettery": "209.09(W)",
  "betteryStatus": 1,
  "betteryStatusStr": null,
  "load": "141.81(W)",
  "loadStatus": -1,
  "grid": "31(W)",
  "soc": 29,
  "socText": "29%",
  "hasEquipment": true,
  "gridStatus": 1,
  "isHomKit": false,
  "isBpuAndInverterNoBattery": false,
  "isMoreBettery": false,
  "genset": "0(W)",
  "gensetStatus": 1,
  "gridGensetStatus": 0
}
```

seems like it is possible to support powerflow data without being HomKit
notice the is `isHomKit: false`

this probably results in the error from HA
```
Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:367
Integration: Sensor (documentation, issues)
First occurred: 10:47:45 AM (2 occurrences)
Last logged: 1:13:29 PM

Error while setting up sems platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 367, in _async_setup_platform
    await asyncio.gather(*pending)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 582, in _async_add_entity
    entity_name = entity.name
                  ^^^^^^^^^^^
  File "/config/custom_components/sems/sensor.py", line 169, in name
    return f"Inverter {self.coordinator.data[self.sn]['name']}"
                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
KeyError: 'name'
```

IsaacInsoll, 2024-02-06:
This sounds like a duplicate of #92 which I have included some "dodgy fix" for if you want to fix it yourself on your machine for now while we wait for the next release which will have a proper fix

IsaacInsoll, 2024-10-08:
I've done a MR that is awaiting approval https://github.com/TimSoethout/goodwe-sems-home-assistant/pull/115

TimSoethout, 2024-10-09:
Fixed by #115 
Released as https://github.com/TimSoethout/goodwe-sems-home-assistant/releases/tag/6.1.0

Otherwise reopen.

closed 2024-10-09 by TimSoethout