TimSoethout/goodwe-sems-home-assistant #100: Energy template in HA incorrect

BrutalDeluxe20, 2024-01-30:
Hello,

I'm having an issue whereby the energy meter in HA is incorrect when utilising this integration for Goodwe SEMS.  I've had it configured for a number of months and it last worked on Christmas Day last year.  Ever since the "grid import" value seems to be logged in HA at midnight from the previous day and causing the days energy use information to be wrong.

**Christmas Day**

[image]

**29th December**
[image]

I've tried the following:
- Updated HA
- Updated this integration
- Completely removed, then re-added this integration and set it all up from scratch
- Rebooted the Goodwe homekit device
- Setup the energy template from scratch

I can login to Goodwe SEMS via the app or web and the data logged there looks correct. 

**Christmas Day**
[image]

**29th December**
[image]

I'm not sure if it's HA or this integration doing something to the data and how it calculates drawing those usage graphs.  I've checked and timezone is correct in HA and I can't see any timezone setting in Goodwe SEMS (besides me putting in my home address where the inverter is installed so presume it sets timezone based off that).

Any help would be really appreciated as this is an awesome integration with HA and I've found the data really useful.

Thanks

TimSoethout, 2024-01-31:
This seems related to #94? Others have found a workaround for it, which you can find there. By disabling update for a while around midnight using automations.

BrutalDeluxe20, 2024-01-31:
Hi @TimSoethout, thanks for linking to that issue, probably would have been good to check if anyone else had reported similar!  Thanks also for your work on this integration as it's so good to have the energy data in HA.

BrutalDeluxe20, 2024-02-02:
Checked #94 and seems to be the same.  I've implemented the workaround in that issue and seems to be working perfectly again.  I'll close this off. Thanks

closed 2024-02-02 by BrutalDeluxe20