TimSoethout/goodwe-sems-home-assistant #104: Max retries exceeded with url: xxx

jirisykora83, 2024-03-26:
Logger: custom_components.sems.sems_api
Source: custom_components/sems/sems_api.py:130
integration: GoodWe SEMS PV API ([documentation](https://github.com/TimSoethout/goodwe-sems-home-assistant), [issues](https://github.com/TimSoethout/goodwe-sems-home-assistant/issues))
First occurred: March 21, 2024 at 7:40:32 PM (510 occurrences)
Last logged: 3:10:32 PM

Unable to fetch data from SEMS. HTTPSConnectionPool(host='eu.semsportal.com', port=443): Max retries exceeded with url: /api//v2/PowerStation/GetMonitorDetailByPowerstationId (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f683707d010>: Failed to establish a new connection: [Errno -3] Try again'))
Unable to fetch data from SEMS. HTTPSConnectionPool(host='eu.semsportal.com', port=443): Max retries exceeded with url: /api//v2/PowerStation/GetMonitorDetailByPowerstationId (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f6837318620>: Failed to establish a new connection: [Errno -3] Try again'))
Unable to fetch data from SEMS. HTTPSConnectionPool(host='eu.semsportal.com', port=443): Max retries exceeded with url: /api//v2/PowerStation/GetMonitorDetailByPowerstationId (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f6836e86f90>: Failed to establish a new connection: [Errno -3] Try again'))
Unable to fetch data from SEMS. HTTPSConnectionPool(host='eu.semsportal.com', port=443): Max retries exceeded with url: /api//v2/PowerStation/GetMonitorDetailByPowerstationId (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f6836f4fe30>: Failed to establish a new connection: [Errno -3] Try again'))
Unable to fetch data from SEMS. HTTPSConnectionPool(host='eu.semsportal.com', port=443): Max retries exceeded with url: /api//v2/PowerStation/GetMonitorDetailByPowerstationId (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f6836ef3e30>: Failed to establish a new connection: [Errno -3] Try again'))

Context:
I have connected more power plant and it seems to not working correctly.

freddy357, 2024-03-30:
Hello,
Same issue here. Since last update.

This error originated from a custom integration.

Logger: custom_components.sems.sems_api
Source: custom_components/sems/sems_api.py:130
integration: GoodWe SEMS PV API (documentation, issues)
First occurred: March 27, 2024 at 16:34:32 (6 occurrences)
Last logged: March 28, 2024 at 12:14:41

Unable to fetch data from SEMS. HTTPSConnectionPool(host='eu.semsportal.com', port=443): Read timed out. (read timeout=30)

This error originated from a custom integration.

Logger: custom_components.sems.sensor
Source: helpers/update_coordinator.py:350
integration: GoodWe SEMS PV API (documentation, issues)
First occurred: March 27, 2024 at 16:34:32 (3 occurrences)
Last logged: March 28, 2024 at 12:11:41

Error fetching SEMS API data: Error communicating with API: 'NoneType' object is not subscriptable

robinhood-code, 2024-04-24:
I had this issue when the SEMS account is still logged in to the SEMS portal and the session has not expired. Once I've logged out the SEMS portal then it was fine.

jirisykora83, 2024-04-24:
> I had this issue when the SEMS account is still logged in to the SEMS portal and the session has not expired. Once I've logged out from the SEMS portal then it was fine.

How long does it take to login expire? I have more Plant so i create multiple account (for each plant) but i do not know which account is use for each integration. There is no way to show configuration after it is set #105

robinhood-code, 2024-04-24:
you can check the config in /.storage/core.config_entries, search for "sems" and match your plant with title / powerstation_id. You'll see your username and password there. 

I have to add that I also disabled the integration for a few minutes after I have logged out from the SEMS portal, before re-enabling it again, and the integration worked after that.