TimSoethout/goodwe-sems-home-assistant #106: Version 5.0.0 is making my GoodWe inverter offline

danwinp, 2024-03-28:
After upgrading from 3.7.4 to 5.0.0 my SEMS portal app and web interface started going offline. When I downgraded to 3.7.4 it has fixed it.

I have tried different intervals too. Longest interval I tried was 10 minutes and it still broke in 5.0.0.

I am currently using 60 seconds interval in 3.7.4 and that seems to work fine.

TimSoethout, 2024-03-28:
I don't have any issues in my log though.
Do you mean the HASS component breaks your regular sems site/app?

danwinp, 2024-03-28:
Yep it makes the sems site/app go offline after a certain time and it won't go back online unless I restart the inverter physically.

danwinp, 2024-03-29:
I installed this using HACS on my home assistant in Truenas Scale.

TimSoethout, 2024-03-29:
> [Tracking image][https://github.com/notifications/beacon/AAEQZ3CQZ6Z7HZJFW3VQPWTY2VUQVA5CNFSM6AAAAABFL4CJ3KWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTY2VS5I.gif]

I will try to look into this if I find some time.
Does it also happen in v4.0.0? In order to deltadebug a bit more.

It sounds a bit like SEMS is rate limiting you. Also you can try using a separate/"non-read" user for the plugin. I think there is some info in the readme on this.

danwinp, 2024-03-29:
I think it was ok with V4.0.0. I had it in V4.0.0 for a little more than a day or so and I don't remember it being an issue. It only started when I upgraded to V5.0.0.

I did make a separate guest/visitor account for it.

danwinp, 2024-03-29:
I just went straight to 3.7.4 as I have been on that for awhile without any issues. I needed my inverter to be working in the next several days as I use the automation to charge my EV car. I can try upgrading to V4 maybe after a week or so and report back to you if there is any issues.

danwinp, 2024-03-29:
Also do you know if there are settings to prevent it from rate limiting me?

TimSoethout, 2024-03-29:
No settings unfortunately. It seems the switch PR (5.0.0) includes some non-optimal code that calls the API twice for each sensor. Maybe that results in you being rate limited. I'm currently coding something up, but will need to test it first. I'll try to do that over the weekend.

TimSoethout, 2024-03-29:
Maybe combining it with #89 and #91.

danwinp, 2024-03-29:
Awesome. No rush, I really appreciate the work you've done for this!

jonathansiero, 2024-09-25:
I'm also getting this. 

Seems to have be happening with version 6.0.0 for me as well.