TimSoethout/goodwe-sems-home-assistant #107: Docs / list attributes

jirisykora83, 2024-03-30:
Would be nice to have docs/list with all the attributes

linusst, 2024-04-10:
|Sensor/attribute name           |Interpretation                  |My comment                         |
|--------------------------------|--------------------------------|-----------------------------------|
|Pac                             |Power from panels to inverter   |Strange name sins it is dc current |
|Total pbattery                  |Power from battery to inverter  |                                   |
|Outputpower                     |Power from inverter to house    |                                   |
|Pmeter                          |Power from house to grid        |                                   |
|Eday/Etotal                     |Energy from panels to inverter  |                                   |
|Output eday/Output etotal       |Energy from inverter to house   |                                   |
|Buy/Total buy                   |Energy from grid to house       |                                   |
|Seller/Total sell               |Energy from house to grid       |                                   |
|EBatteryCharge/EChargeDay       |Energy from inverter to battery | For me this the total is 1,731.8, not sure someone plase confirm |
|EBatteryDischarge/EDischargeDay |Energy from battery to inverter | For me this the total is 2,344.5, not sure someone plase confirm |
|ETotalBuy/EDayBuy               |Energy from house to inverter   | For me this the total is 1,327.4, so not sure if this is correct but my smart meter was wrongly connected for the first week|

jirisykora83, 2024-05-21:
| Sensor/attribute name  | Interpretation | My comment |
| ------------- | ------------- | ------------- | 
| vbattery1 | Battery Voltage  | |
| ibattery1  | Battery Current  | |
| pbattery1 | Battery Power  | |
| vgrid |  On-grid Voltage  | |
| igrid| On-grid Current  | |
| pgrid | On-grid Export Power  | |

base on: https://github.com/marcelblijleven/goodwe/issues/20

TimSoethout, 2025-01-15:
Feel free to PR this information into the Readme. :)