TimSoethout/goodwe-sems-home-assistant #109: Import data inconsistent

justinmaiuto, 2024-06-15:
Energy import consistency seems to have degraded over the last few days. Screenshot shows typical data capture for the last week, but on the 13/06 something looks to have changed and import readings are low and not linear as expected. This is not reflective of actual consumption, verified by other power monitoring device and also in the SEMS portal/app itself both showing data as expected. Could be an issue on the SEMS server/API side, but putting it here in case it could be something on the data feed/reads on HA side that changed?

[image]

skreza, 2024-06-16:
I'm seeing this same issue, however looking at data in SEMS web portal looks correct.
It's almost like it's failing to connect to SEMS to pull the data in.

[image]

TimSoethout, 2024-06-16:
It seems to reset at midnight. Maybe some auto increment setting in the component HA is missing?
Unfortunately I can't reproduce, since my inverter does not import.

rmartinsgr, 2024-06-16:
Hi,
Same issue here.
The issue has to be in the SEMS API because even there, at the end of the day you can't see the total buy amount of Kwh, only in the next day.
But it shows the realtime consumption, it's strange and in my case it stopped working on 10/06/2024 @23:00..
Don't know what to do next, a lot of automations depend on this because Homekit hk1000 only gives information by cloud API...

[image]

justinmaiuto, 2024-06-16:
> It seems to reset at midnight. Maybe some auto increment setting in the component HA is missing? Unfortunately I can't reproduce, since my inverter does not import.

Resetting at midnight is expected. It's the incorrect/inconsistent data feed/capture you can see from the 13/06 that is the problem it seems

davidsonimagerygmailcom, 2024-06-17:
I am seeing the same issue ... kwh metrics are all skeweed and frozen ... until a day or two later (or sometiunes much later in the day) it suddenly catches up.  I can see the issue happening in Sems portal too, it's not HA.

It all started around 13/14th June.
I"ve raised a case with my solar installer, that's gone nowhere. 

[image]

Can gaurantee that overnight sometime it will magically update (and goodwe will point to that saying "all looks fine")... but all the wrong data will be ingested by HA then.

davidsonimagerygmailcom, 2024-06-17:
> I'm seeing this same issue, however looking at data in SEMS web portal looks correct. It's almost like it's failing to connect to SEMS to pull the data in. [...]

Nah the data is wrong in Sems too ... untill 12-24hrs later, then suddenly it backdates the earlier days to correct figures.

I wish my home had only used 0.5kwh by the time the run rises in morning here winter! :)

davidsonimagerygmailcom, 2024-06-18:
Issue persists.  Apparently the issue is with the way the HK1000 (goodwe smart meter) and SemsPortal are talking.  Ergo, it's not an acute issue but a cloud / semsportal one.

justinmaiuto, 2024-06-18:
> Issue persists. Apparently the issue is with the way the HK1000 (goodwe smart meter) and SemsPortal are talking. Ergo, it's not an acute issue but a cloud / semsportal one.

do you have a ticker or something we could track with GoodWe on this?

davidsonimagerygmailcom, 2024-06-18:
Not really, I had raised a case with my Solar Provider here in WA (who are great) , they had logged a case with Goodwe.  Goodwe had said (after 5 days) that there's no issue (i've since seen screenshot of the conversation, the issue is goodwe not comprehending what was even written).   I've provided further screenshots and now they advised "_We are facing Eotal problem with HK1000.  Our Sems Team is trying their best to resolve this problem.   Once the bug fix we will let you know"_

At no point has anyone given me any ticket number, my solar provider is just emailing with them (under their vendor relationship I imagine).

I can see some guys on Facebook (croatia) reporting similiar.
What I do know is that when the midnight rollover happens, the previous days metrics magically are OK ... that's of little use to us.

I also make a point of not mentioning HA or pulling data into other sources because they straight out don't care or comprehend that (hence the double dip at midnight thing they couldn't comprehend earlier this year) ... or worse, they blame it.

davidsonimagerygmailcom, 2024-06-19:
Today's data looking good.  Haven't said anything back to vendor though , let's see how a couple days goes.

rmartinsgr, 2024-06-19:
> days goes.

Same, seems to be back to normal

TimSoethout, 2024-06-19:
Hooray. I will close the issue then. Please re-open if we need to do anything on this again.

closed 2024-06-19 by TimSoethout

AgentKith, 2024-06-20:
> Hooray. I will close the issue then. Please re-open if we need to do anything on this again.

Side question. Assuming if the WeWork API supports this, is it possible to introduce a HA service call to refresh retrospective data so they're accurate? i.e. go into the cloud and update the wrong days (while they're still being kept).