TimSoethout/goodwe-sems-home-assistant #111: "no version key" after upgrade Core to 2024.8.2

tuxick, 2024-08-25:
Since upgrade i'm getting "The custom integration 'sems' does not have a version key in the manifest file and was blocked from loading." 

**sems/manifest.json definitely has version key:**
```
{
  "domain": "sems",
  "name": "GoodWe SEMS API",
  "config_flow": true,
  "documentation": "https://github.com/TimSoethout/goodwe-sems-home-assistant",
  "issue_tracker": "https://github.com/TimSoethout/goodwe-sems-home-assistant/issues",
  "iot_class": "cloud_polling",
  "requirements": [],
  "dependencies": [],
  "codeowners": ["@TimSoethout"],
  "version": "5.0.0"
}
```
so what is wrong?

TimSoethout, 2024-08-25:
Unsure. For me it works in HA 2024.8.3.
Are you sure the files on your filesystem also contain the version?
Maybe try reinstalling via HACS.

TimSoethout, 2024-08-25:
I only see this warning, which needs to be fixed somewhere this year:

```
2024-08-25 20:09:25.150 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'sems' calls async_forward_entry_setup for integration, sems with title:
989ea198-7dd9-43f8-80c1-ea366808d942 and entry_id: d787c6ed2def272f275c27e6bab157c3, which is deprecated and will stop working in Home Assistant 2025.6, await async_forward_entry_se
tups instead at custom_components/sems/__init__.py, line 37: hass.async_create_task(, please create a bug report at https://github.com/TimSoethout/goodwe-sems-home-assistant/issues
```

tuxick, 2024-08-26:
After restarting i got the same message again but getting data anyway. Weird.

kaizersoje, 2024-08-28:
I get this warning

`
2024-08-27 17:40:53.272 WARNING (MainThread) [homeassistant.helpers.frame] Detected code that calls async_forward_entry_setup for integration sems with title: [..] and entry_id: [..], during setup without awaiting async_forward_entry_setup, which can cause the setup lock to be released before the setup is done. This will stop working in Home Assistant 2025.1. Please report this issue.
`

closed 2024-09-18 by TimSoethout