TimSoethout/goodwe-sems-home-assistant #114: Can't access API

Prana66, 2024-09-17:
I'm trying to log in, but I get an authentication error, the username and password are correct because I log in to the website with them.
The Power Station ID, I have followed your instructions by putting the number after https://www.semsportal.com/PowerStation/PowerStatusSnMin/
But I can't connect.

Prana66, 2024-09-17:
Now I have been able to enter but with the old password (not the same as the current one) which is kept for the integration, but without devices or entities.
[image]

TimSoethout, 2024-09-18:
Really odd. Do the debug logs say anything? (See readme)
Maybe you created a separate user that is not allowed to see the inverter?

Prana66, 2024-09-18:
Hello, thank you for answering, I can't send you anything because there is no device or integration, I don't know what I could send you, as you can see in the screenshot there is nothing.
In the debug log:
2024-09-17 13:57:49.698 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'sems' calls async_forward_entry_setup for integration, sems with title: 95000______8 and entry_id: 01J7R___________________S9R1ZQ, which is deprecated and will stop working in Home Assistant 2025.6, await async_forward_entry_setups instead at custom_components/sems/__init__.py, line 37: hass.async_create_task(, please create a bug report at https://github.com/TimSoethout/goodwe-sems-home-assistant/issues

TimSoethout, 2024-09-19:
Above WARNING line should be solved in the latest release. But is probably not related to your issue.

See here for enabling debug logging: https://github.com/TimSoethout/goodwe-sems-home-assistant?tab=readme-ov-file#debug-info

Prana66, 2024-09-19:
These are the three records that appear, I still think that the problem is that it does not update, it asks me for the password from a year ago.

Este error se originó a partir de una integración personalizada.

Registrador: custom_components.sems.sems_api
Fuente: custom_components/sems/sems_api.py:130
integración: GoodWe SEMS API (documentación, problemas)
Ocurrió por primera vez: 00:43:39 (8 ocurrencias)
Última vez registrado: 10:13:02

Unable to fetch data from SEMS.
-------------------------------------------------------------------------------------------------------------------------------

registrador: homeassistant.components.goodwe.coordinatorFuente: helpers/update_coordinator.py:386integraci%C3%B3n: GoodWe Inverter (documentaci%C3%B3n, problemas)Ocurri%C3%B3 por primera vez: 02:26:01 (1 ocurrencias)%C3%9Altima vez registrado: 02:26:01Error fetching GoodWe data: ('No valid response received even after 10 retries', 3)
---------------------------------------------------------------------------------------------------------------------------
Registrador: homeassistant.helpers.frame
Fuente: helpers/frame.py:184
Ocurrió por primera vez: 00:43:37 (1 ocurrencias)
Última vez registrado: 00:43:37

Detected that custom integration 'sems' calls async_forward_entry_setup for integration, sems with title: 95000EH27W0028 and entry_id: 01J7RTMP5JNW_____-8S9R1ZQ, which is deprecated and will stop working in Home Assistant 2025.6, await async_forward_entry_setups instead at custom_components/sems/__init__.py, line 37: hass.async_create_task(, please create a bug report at https://github.com/TimSoethout/goodwe-sems-home-assistant/issues

DannyMake, 2024-10-14:
I got the same problem.

BenBarat, 2024-12-07:
First: 'Terms of Use' tickbox on the website needs to be checked before login can occur, hence authentication **will consistently fail** if just the UserName and Password are supplied when screen scraping.

Second: I note the URL (that determines the ID) now **changes each time** the website is accessed. This makes the premise the ID can be stored locally on HomeAssistant configuration files a false one. Going back even further in the docs, it raises the question of exactly where does the login and password information get stored if you use a file editor to hard code the credentials?

These are GoodWe changes that have cropped up since this code was deployed, possibly as a result of exploring alternate API objectives, or tweaking their website. Their 'Solar Academy' website also has a lot of information about APIs in an August 2024 paper, listing their intentions, but is actually quite light on implementation and sample code. It mentions licensing agreements, etc. It is about this time I suspect the problem arose, and some users that have cached credentials are living on borrowed time.

Maybe there are now alternate URL's or portals to the data? A JSON portal? HTTPS protocols and KAFKA? OpenAPI? Real-time Data Monitoring API? Is contacting GoodWe on behalf of the community going to get a specific agreement to make this mod work again?

As a comment, having a screen pop up in the initial load of the integration that asks for the User Name, Password, and then ??? may be looking at this wrong. Shouldn't you login in, ascertain the URL, and then update the configuration file (or store that value in a temporary variable) with the correct ID that has now been offered up by the GoodWe website?

This screen scraper mod is **firmly broken** till that authentication issue is remedied.

Ascariota, 2025-02-24:
Same issue for me

Prana66, 2025-02-24:
I'm solved,

Ascariota, 2025-02-24:
@Prana66 : 
how I don't see the solution

Thanks

Prana66, 2025-02-24:
Well, with patience, it was solved by itself, I don't know if it was due to
some update

Ascariota, 2025-02-24:
I don't know if it's help.

But I delete the fold older Sems integration on Custom_component, and I reset the passwors SEMS portal with exactly the same password. 

And after a reboot it's works !

BenBarat, 2025-05-02:
> I'm not sure about @BenBarat's story. This component uses the API, and it not screen scraping or anything. For me the inverter ID, which is copied from the URL, seems stable .
> (I seems his Github account was created for posting above message only? No other activity on that account.)
> 

I was a little taken aback at your comment, which may have invoked a feeling of being treated as a malicious spammer. We all have a first post - you were privileged to be one of the first! I'm not sure if you intended to attack the messenger while being delivered the message, right or wrong. Yes, after being a avid reader of open source documentation and following installation steps closely, resulting in a less than stellar experience with a lot _(most)_ of the Home Assistant GoodWe add-ons not working, I did sign up on GitHub to specifically discuss the issues I believe were needing a little attention from developers to make a more robust platform. No, I am not a newcomer to the computer world, having cut my teeth on punch cards as a kid, and getting $600 per hour while consulting during the Y2K situation, and enjoying automating my house to an inch of its' life. Not sure how much comments are expected on GitHub, but I'm trying my best to keep a low profile and not pollute the Internetz with idle chit-chat and sketchy code. Having been involved in tech support of various kinds over the decades, including being moderator of a number of robust high-volume technical forums, I have observed that volume of content is not always directly related to quality of content. I also note developers that address user feedback seem to be the more successful ones.

"Web scrape" - maybe I was misled when reading - it says so in your installation instructions:

> **Manual Setup**
> **Crude** sensor for Home Assistant that **scrapes from GoodWe SEMS portal**. Copy all the files in custom_components/sems/ to custom_components/sems/ your Home Assistant config dir.

Based on that comment, I invoked the _examine source code_ option on my browser to see what was being offered up for scraping. I note other GoodWe add-ons attempt to web scrape with varying success, and the current Home Assistant built-in implementation did not work with my older three phase DT series inverter. Disappointingly, various suggestions were offered in the discussions for this, including updating outdated firmware and replacing functional hardware, and the threads are closed without resolving the issue. I also note that some of this code did actually work with DT series inverters at some point and that something had changed, either in their code, the underlying web of dependencies with other open-source code and subroutines, or at the GoodWe end, whether a firmware update, web portal update, or change to the ‘mysterious API’. Of course, the SEMS portal has been consistently and reliably been updated over the years with valid data by my inverter, so I take those comments with a little helping of salt. If it ain't broke, don't fix it... If something has changed, let’s find out why and act accordingly, not throw our hands in the air and discard a working environment.

Further on in the same README.md document under the title "Extra (optional) templates to easy access data as sensors" it speaks of adding some **templates**. Where to add and how? Could this be a worthy addition to the initialization subroutine automagically so that everything just works well out of the box? A hassle free add-on that just works? Yeah/naah...

The instructions in the following section:

> **Configure integration**
> 
> The **required ID of your Power Station** can be retrieved by logging in to the SEMS Portal with your credentials: https://www.semsportal.com
> 
> After login you'll see the ID in your URL, e.g.: https://semsportal.com/PowerStation/PowerStatusSnMin/12345678-1234-1234-1234-123456789012
> 
> In this example the ID of the Power Station is: 12345678-1234-1234-1234-123456789012
> 
> In the home assistant GUI, go to Configuration > Integrations and click the Add Integration button. Search for GoodWe SEMS API.
> 
> Fill in the required configuration and it should find your inverters.
> 
> Note that change**d** to configuration.yaml [b]are no longer necessary[/b] and can be removed.[/quote]

 (my emphasis)

Should that be 'changes' in the last line? A little typo? I'm happy an example has been given, as I am used to following and standing on the shoulders of giants.

Where it talks later about adding templates it says 'Replace $NAME with your inverter entity id.' - is this $NAME the ID of the PowerStation mentioned earlier or something else, and can you add that as a further example to clarify the documentation?

I understand open-source is a labor of love, and admire your efforts, but please devote time and focus to your code, not denigrating newcomers. I apologize if I am a bit touchy - your comment seemed a little harsh.

I observe the GoodWe authentication process your code follows seems to involve a (1)username (email address), (2)password, and (3)access token, which I note you have modified your code to verify, and if necessary, update. I an guessing this access token being (a) wrong, (b) being 'accidentally' right due to caching and then the cache expiring, or (c) just plain overlooked, it is the underlying fatal flaw in the other GoodWe access software that has been attempted over the last few years. The other comments in this thread seems to confirm that, including on [... cut ...] bling block in my installation process. Please advise what needs to go in there, where and how to find it, and how to do it, and update the README.md file, and your offering will probably become the component of choice for HomeAssistant GoodWe owners. If you are able to also automate that process to prevent typing and pasting errors, all the better.

Thank you for your contributions, and looking forward to getting some useful statistics out of my elderly but reliable GoodWe DT inverter. 
Cheers.