TimSoethout/goodwe-sems-home-assistant #120: entities Goodwe Unavailable

snhnic, 2025-01-10:
Reload seems okay, but I don't have entities anymre? 

Inverter: This entity is no longer being provided by the sems integration. If the entity is no longer in use, delete it in settings.
Goodwe inverter Energy  This entity is no longer being provided by the sems integration. If the entity is no longer in use, delete it in settings.

Tonny56, 2025-01-14:
I have the same problem.

TimSoethout, 2025-01-14:
> [Tracking image][https://github.com/notifications/beacon/AAEQZ3BH2K4YDUA5BAMT2RL2KTH77A5CNFSM6AAAAABU6AEISCWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTU2K3MZ2.gif]

Feel free to check the logs, also on debug. See readme for instructions

snhnic, 2025-01-14:
what good would that do? It worked fine till it did not. I changed nothing. So api change or...

TimSoethout, 2025-01-14:
It could provide insight on what is wrong.

Tonny56, 2025-01-14:
I just reloaded the integration and it works fine again!

xEricxD, 2025-01-14:
I'm having a similar issue. It logs the following error which seems to be related:
`
2025-01-14 17:19:13.066 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2349, in async_forward_entry_setup
    result = await self._async_forward_entry_setup(entry, domain, True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2380, in _async_forward_entry_setup
    await integration.async_get_platform(domain)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1108, in async_get_platform
    platforms = await self.async_get_platforms((platform_name,))
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1185, in async_get_platforms
    import_future.result()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1173, in async_get_platforms
    platforms.update(self._load_platforms(platform_names))
                     ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1098, in _load_platforms
    platform_name: self._load_platform(platform_name)
                   ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1272, in _load_platform
    cache[full_name] = self._import_platform(platform_name)
                       ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1304, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/sems/sensor.py", line 19, in <module>
    from homeassistant.components.sensor import STATE_CLASS_TOTAL_INCREASING, SensorEntity
ImportError: cannot import name 'STATE_CLASS_TOTAL_INCREASING' from 'homeassistant.components.sensor' (/usr/src/homeassistant/homeassistant/components/sensor/__init__.py)
`

Does that help at all, or is there other info you need?

TimSoethout, 2025-01-15:
You're using an older version. This was fixed in https://github.com/TimSoethout/goodwe-sems-home-assistant/pull/102 .
Please upgrade to the latest version (via HACS).
Good luck!

closed 2025-01-15 by TimSoethout

snhnic, 2025-01-15:
closed???? But my problem has not been solved!!!! What log do you need? 
I deleted the integration and haave started over, but no entities.

[image]

TimSoethout, 2025-01-15:
Glad it is solved. I think HACS might use an older version if you explicitly selected it when installing. Weird indeed.