TimSoethout/goodwe-sems-home-assistant #121: GoodWe HomeKit Exported and Imported values freeze frequently while the Power value and entities remain live

RobSmyth, 2025-02-28:
The GoodWe Smart Meter (HomeKit) device's imported and exported entities frequently flat line for a few hours although the "Smart Meter" entity (power) remains live. All attributes on the power entity are also live and the exported and imported values can be calculated from these attributes.

The GoodWe Smart Meter (HomeKit) device has 3 entities:

- Smart meter (power) with many attributes.
- Exported
- Imported

[image]

The history below shows the Exported and Imported entities often flat lining while the power entity remain live.

[image]

This resulted in the energy card sometimes showing massive house consumption, no export data, or massive values in random hours.

The interesting thing is when I reconfigured the energy card to use entities created from templates and helpers, instead of the integration's Exported and Imported entities, energy card is now (so far) worked well. The integration's entity's entities still flat line but the power entity and its attributes are reliable.

This may explain some of the problems other users have seen with bumps and apparent lock-ups.

TimSoethout, 2025-02-28:
Interesting. Does that mean the import/export values also "flatline" in the api calls? Or do they fail maybe?

RobSmyth, 2025-02-28:
> Interesting. Does that mean the import/export values also "flatline" in the api calls? Or do they fail maybe?

I have not checked the logs as yet. I will get back to you on that. Do the import and export values come from SEMS or derived from the power value attributes? I ask as I get perfect import and export values using the 1st entity's attributes.

RobSmyth, 2025-02-28:
Hi @TimSoethout 

Here is more info taken this morning.

The 3 smart meter entities showing 2 having odd values and flat lined.:

[image]

Redacted and partially formatted log attached:

[Logs-redacted.txt](https://github.com/user-attachments/files/19034238/Logs-redacted.txt)

RobSmyth, 2025-02-28:
@TimSoethout 

FYI - I do not poll from 11:55 pm to 00:15pm.

RobSmyth, 2025-03-01:
@TimSoethout 

I've had a chance now to look at the logs. This is first time I've looked at the SEMS response and my guesses are:

1. SEMS is reponding
2. The import and export values are in the `energeStatisticsCharts.buy` and `energeStatisticsCharts.sell` JSON parameter (shown below) and are incorrect at the time I took the logs and screen captures.

```json
    "hasEnergeStatisticsCharts": "True",
    "energeStatisticsCharts": {
        "contributingRate": 0.65,
        "selfUseRate": 0.28,
        "sum": 37.57,
        "buy": 5.74,
        "buyPercent": 35.1,
        "sell": 26.97,
        "sellPercent": 71.8,
        "selfUseOfPv": 10.6,
        "consumptionOfLoad": 16.34,
        "chartsType": 1,
        "hasPv": "True",
        "hasCharge": "False",
        "charge": 0.0,
        "disCharge": 0.0,
        "gensetGen": 0.0,
        "hasGenset": "False",
        "isStored": "False"
    },
```

The really odd thing is that the power consumption entity has correct value and attributes allowing the correct day import and export energies to be calculated.

I've not looked at the integration's code yet.

RobSmyth, 2025-03-02:
Looking back a the Export energy entity history from Jan 2023 I can see that it did flat line and/or not reset fully to zero overnight about once a month up to October 6 2024 when it was useless most days. See chart below.

[image]

I know that real time power data, which I calculate from the smart meter's power value attributes, has been good as I do look at it every day over the last 2 years, so I do not think it has been off-line.

TimSoethout, 2025-03-02:
> [@TimSoethout](https://github.com/TimSoethout)
> 
> I've had a chance now to look at the logs. This is first time I've looked at the SEMS response and my guesses are:
> 
>     1. SEMS is reponding
> 
>     2. The import and export values are in the `energeStatisticsCharts.buy` and `energeStatisticsCharts.sell` JSON parameter (shown below) and are incorrect at the time I took the logs and screen captures.
> 
> 
>     "hasEnergeStatisticsCharts": "True",
>     "energeStatisticsCharts": {
>         "contributingRate": 0.65,
>         "selfUseRate": 0.28,
>         "sum": 37.57,
>         "buy": 5.74,
>         "buyPercent": 35.1,
>         "sell": 26.97,
>         "sellPercent": 71.8,
>         "selfUseOfPv": 10.6,
>         "consumptionOfLoad": 16.34,
>         "chartsType": 1,
>         "hasPv": "True",
>         "hasCharge": "False",
>         "charge": 0.0,
>         "disCharge": 0.0,
>         "gensetGen": 0.0,
>         "hasGenset": "False",
>         "isStored": "False"
>     },
> 
> The really odd thing is that the power consumption entity has correct value and attributes allowing the correct day import and export energies to be calculated.
> 
> I've not looked at the integration's code yet.

So, If I understand correctly the data is both wrong in the response JSON and in HA (graphs)?

It could be that somehow browsing the website forces the backend to update the `energeStatisticsCharts` statistics or something. Just a hunch.

Does it make sense to take a different value from the JSON? Maybe from `energeStatisticsTotals`? Or indeed do the calculations you do in the component to compensate?

RobSmyth, 2025-03-02:
Hi @TimSoethout ,

> So, If I understand correctly the data is both wrong in the response JSON and in HA (graphs)?

Correct, both are often wrong when I use the Import and Export entities in the energy dashboard.

> It could be that somehow browsing the website forces the backend to update the energeStatisticsCharts statistics or something. Just a hunch.

FYI - I do not remember using the SEMS web site for a couple of years now. I do not use it.

> Does it make sense to take a different value from the JSON? 

Yes, the power entity and its attributes are correct and can be used to determine energy consumed.

IMO - The Import and Export entities on the HomeKit integration are not usable as the SEMS data is unreliable. They often show incorrect data. They need to be removed or changed.

> Or indeed do the calculations you do in the component to compensate?

The power entity and its attributes have been reliable over the last couple of years and I have good energy and power charts using them. Sufficiently accurate for the purpose of home power consumption. This also fixes the energy data wobble after midnight issue.

What works well is:

1. Get real time Importing and Exporting power from the power entity attributes. (Uses `loadStatus` attribute)

   This also add power Importing and Exporting for direct for use in power chart like this...

[image]

2. Integrate these to get correct Imported and Exported energy values. 

I can provide my templates and helpers if you like. Nothing really new but I can give and end to end.
I think this could all be done within the integration. That would be very nice.

**Also ...**

SEMS API has multiple issues that have similar symptoms (flat line). Ones I know of:

- 7 minutes after midnight energy data wobble
- SEMS can take hours to allow a re-connection (this causes both real-time power and historical energy to flat line)
- SEMS day energy data are unreliable (this issue)
  - Flat lines for hours
  - Does not reliably reset to zero daily

So an On-line, or device state, entity would be useful to make state clear to users. Or ... much better, return null values so that gaps are shown on the power chart to show connection lost (no flat line back fill on connection).

Note: I poll every minute have not seen an issue with polling rates.

RobSmyth, 2025-03-02:
It would be great if the HomeKit device entities were:

```
States:

  Power           // same as existing, nice if null when not connected
  Importing     // >0 if importing
  Exporting     // >0 if exporting
  Consuming  // >0 if consuming locally  (load attribute)

  Imported     // today's accumulated imported energy
  Exported     // today's accumulated exported energy

Diagnostics:

  State          // device connected state
  
```

Perhaps make `Imported` and `Exported` configurable to be calculated in the integration or come from SEMS.

TimSoethout, 2025-03-03:
Great ideas. Thanks for looking into it.
I currently don't have the time/energy to make big changes to the integration, but can review any PRs that come in.

RobSmyth, 2025-03-05:
Hi @TimSoethout 

I'm looking at the code to see if I can help with the above code change. As I browse the code I see non-related details I would like to ask about ... what is the best way to chat? Would enabling discussions on the GitHub project be good for that?

I do not want to pollute discussion on this issue but FYI - one thing I want to ask was if the `SensorDeviceClass.POWER_FACTOR` should be `SensorDeviceClass.POWER`?

```python
    @property
    def device_class(self):
        return SensorDeviceClass.POWER_FACTOR
``` 

I've coded in Python but I'm a novice. It is not my primary coding language so I'm learning how HASS integrations are done :-)

TimSoethout, 2025-03-05:
Discussions sounds useful. It should be enabled now. 

You might very well be correct that we should use `POWER`. Also some additions to the integration (especially the HomeKit part) are contributions of others. I have a WIP branch where I'm completely overhauling/streamlining the code base, but no time to finish it.

If you're have experience coding Python should not be a problem for you. Also in HA programming it is mainly following the code style the HA API gives you in my experience.

TimSoethout, 2026-02-07:
Then I'll close it for now. Feel free to reopen, or open a new issue.

closed 2026-02-07 by TimSoethout