TimSoethout/goodwe-sems-home-assistant #126: Cannot see Grid Voltage?

Sage222, 2025-05-15:
Hi - is there anyway to see Grid Voltage?  It is listed on the SEMS portal - though i cant see it in your plugin? thanks very much

TimSoethout, 2025-05-15:
> [Tracking image][https://github.com/notifications/beacon/AAEQZ3FD4TR6CHTUN2RRVIT26Q7VTA5CNFSM6AAAAAB5FKDYPSWGG33NNVSW45C7OR4XAZNFJFZXG5LFVJRW63LNMVXHIX3JMTHLNNBSBU.gif]

Hi, check the attributes of the power sensor.

In a future update I plan to make it a separate sensor. For now you can create a template sensor, see the readme.

Sage222, 2025-05-15:
thanks for the response will check it out!

TimSoethout, 2025-06-19:
> still not available on the 6.1.1. Should i try the 7.2 beta? only options come up on 6.1.1 is the following:
>
> image.png (view on web)[https://github.com/user-attachments/assets/a00d53b0-22b7-4130-a77f-8ce9a11ea202]

Indeed, you can try 7.2.0 nonbeta.

Sage222, 2025-06-22:
not that i can see.. only the entities above.  though its reported on the SEMS portala and the app .. so it is available.

[image]

TimSoethout, 2025-06-22:
The entity should contain attributes such as:
https://github.com/TimSoethout/goodwe-sems-home-assistant/raw/master/images/sems-details.webp
from the readme.

VidarParry, 2025-12-04:
@Sage222 Thank you. This is what I was hoping to extract as well.

However, I can't get the sensor to appear using your code.

In configuration.yaml I'm using:
```
sensor:
  - name: "Inverter Voltage"
    unique_id: inverter_voltage_sensor
    unit_of_measurement: "V"
    state: "{{ state_attr('sensor.inverter_XXXXXXXXX', 'vac1') }}"
```

I can't see the vac1 sensor in the available entities for the inverter, nor can I add it (vac1).

Appreciate any guidance (sorry - I'm new to HA and YAML).
Cheers...Steve

Sage222, 2025-12-04:
use chatgpt it will step you through it. essentially need to go to Developer Settings > States > and find the Inverter device and Attribute/Property that you want.  Then you need to edit your configuration.yaml file to add a new Sensor for that attribute.  Then reboot HA and it should appear.  Chatgpt will step you through it better than i can :)

VidarParry, 2025-12-05:
Thanks Sage222. ChatGPT nor CoPilot helped - I was following that process already.

Found the issue - in my setup (don't know why) the vac1 is an attribute of sensor.inverter_XXXXXXXXXXX_power. So what I ended up with was:

```
template:
  - sensor:
    - name: "SEMS Inverter Grid Voltage"
      unique_id: sems_inverter_grid_voltage_sensor
      unit_of_measurement: "V"
      state: "{{ state_attr('sensor.inverter_XXXXXXXXXXX_power', 'vac1') }}"
```

All working now.

Cheers
...Steve