TimSoethout/goodwe-sems-home-assistant #133: 7.2.0 broken

justinmaiuto, 2025-06-19:
Upgraded to 7.2.0 and some sensors failed to setup. Downgraded back to 6.1.1 (previously working) and now that is broken too.

Inverter power and Homekit Energy are now 'unavailable' in HA.

This is what is in the logs:

```
2025-06-20 09:44:47.124 ERROR (MainThread) [homeassistant.components.sensor] Error adding entity None for domain sensor with platform sems
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 657, in _async_add_entities
    await self._async_add_entity(
        entity, False, entity_registry, config_subentry_id
    )
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 796, in _async_add_entity
    entity_name = entity.name
                  ^^^^^^^^^^^
  File "/config/custom_components/sems/sensor.py", line 177, in name
    return f"Inverter {self.coordinator.data[self.sn]['name']}"
                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
KeyError: 'name'
2025-06-20 09:44:47.126 ERROR (MainThread) [homeassistant.components.sensor] Error adding entity None for domain sensor with platform sems
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 657, in _async_add_entities
    await self._async_add_entity(
        entity, False, entity_registry, config_subentry_id
    )
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 796, in _async_add_entity
    entity_name = entity.name
                  ^^^^^^^^^^^
  File "/config/custom_components/sems/sensor.py", line 273, in name
    return f"Inverter {self.coordinator.data[self.sn]['name']} Energy"
                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
KeyError: 'name'
```

TimSoethout, 2025-06-20:
Yes, 7.2.0 migrated to some new entity ids automatically. Reinstalling the old version then points to the old ones. 

Backup indeed should work.
I guess the new version was not supporting HomeKit properly.

TimSoethout, 2025-06-20:
Can you try `7.3.0-beta` with extra debug logging around etotal?
Not sure what is going on, and hard for me to debug since I don't have a HomeKit.

justinmaiuto, 2025-06-20:
Just tried 7.3.0 beta

```
Error adding entity None for domain sensor with platform sems
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 657, in _async_add_entities
    await self._async_add_entity(
        entity, False, entity_registry, config_subentry_id
    )
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 828, in _async_add_entity
    if self.config_entry and (device_info := entity.device_info):
                                             ^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/sems/sensor.py", line 252, in device_info
    name=f"Inverter {self.coordinator.data[self.sn]['name']}",
                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
KeyError: 'name'
```

```
Error adding entity sensor.solar_smart_meter_sems_homekit_energy for domain sensor with platform sems
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 657, in _async_add_entities
    await self._async_add_entity(
        entity, False, entity_registry, config_subentry_id
    )
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 979, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1370, in add_to_platform_finish
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1019, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1141, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1081, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1025, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/config/custom_components/sems/sensor.py", line 309, in state
    return data["etotal"]
           ~~~~^^^^^^^^^^
KeyError: 'etotal'
```

`Skip migration of id [5010KMST22BW0458] to [5010KMST22BW0458-power] because it already exists`

```
Error adding entity None for domain sensor with platform sems
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 657, in _async_add_entities
    await self._async_add_entity(
        entity, False, entity_registry, config_subentry_id
    )
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 828, in _async_add_entity
    if self.config_entry and (device_info := entity.device_info):
                                             ^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/sems/sensor.py", line 252, in device_info
    name=f"Inverter {self.coordinator.data[self.sn]['name']}",
                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
KeyError: 'name'
```

```
Logger: homeassistant.helpers.event
Source: helpers/template.py:646
First occurred: 07:08:56 (2 occurrences)
Last logged: 07:08:56
[... 33 lines cut ...]

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 761, in async_render_to_info
    render_info._result = self.async_render(  # noqa: SLF001
                          ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
        variables, strict=strict, log_fn=log_fn, **kwargs
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 646, in async_render
    raise TemplateError(err) from err
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ states ('sensor.inverter_5010kmst22bw0458') | float | round(0)
 - states ('sensor.homekit_91000hku22a50161') | float | round(0)}}' but no default was specified
```

```
Logger: homeassistant.components.template.template_entity
Source: components/template/template_entity.py:215
integration: Template (documentation, issues)
First occurred: 07:08:56 (2 occurrences)
Last logged: 07:08:56

TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ states ('sensor.inverter_5010kmst22bw0458') | float | round(0) - states ('sensor.homekit_91000hku22a50161') | float | round(0)}}' but no default was specified') while processing template 'Template<template=({{ states ('sensor.inverter_5010kmst22bw0458') | float | round(0) - states ('sensor.homekit_91000hku22a50161') | float | round(0)}}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.current_grid_export_import'
TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{% if states('sensor.current_grid_export_import') | float | round(0) >= 0 %} {{ states('sensor.inverter_5010kmst22bw0458') | float | round(0) - states('sensor.current_grid_export_import') | float | round(0) }} {% elif states('sensor.current_grid_export_import') | float | round(0) < 0 %} {{ states('sensor.homekit_91000hku22a50161') | float | round(0) + states('sensor.current_grid_export_import') | float | round(0) }} {% endif %}' but no default was specified') while processing template 'Template<template=({% if states('sensor.current_grid_export_import') | float | round(0) >= 0 %} {{ states('sensor.inverter_5010kmst22bw0458') | float | round(0) - states('sensor.current_grid_export_import') | float | round(0) }} {% elif states('sensor.current_grid_export_import') | float | round(0) < 0 %} {{ states('sensor.homekit_91000hku22a50161') | float | round(0) + states('sensor.current_grid_export_import') | float | round(0) }} {% endif %}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.current_solar_self_consumption
```'

wifievangelist, 2025-06-21:
Can confirm it is still broken on 7.3.0. Have reverted back to 6.1.1 to make it work again.

daveedwards123, 2025-06-22:
Using 7.3.0-beta i get the same, i hope these logs help:

`Logger: homeassistant.components.template.template_entity
Source: components/template/template_entity.py:215
integration: Template (documentation, issues)
First occurred: 5:06:31 PM (3 occurrences)
Last logged: 5:06:31 PM

TemplateError('ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.inverter_inverter') | replace(",", "") | float) - (states('sensor.pv_excess') | replace(",", "") | float) }}' but no default was specified') while processing template 'Template<template=({{ (states('sensor.inverter_inverter') | replace(",", "") | float) - (states('sensor.pv_excess') | replace(",", "") | float) }}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.house_electricity_usage'
TemplateError('UndefinedError: None has no element 0') while processing template 'Template<template=({% set prefix='https://edwardshomeassist.duckdns.org/api/hassio_ingress/3-NJxerzOzNSyTRa3APRHLRxNW9T23LPMa095OjeME4' %} {% set url=state_attr('sensor.grocy_meal_plan','meals')[0].picture_url | replace("/grocy/", "/files/",1) %} {% set suffix='?force_serve_as=picture&best_fit_width=400' %} {% if state_attr('sensor.grocy_meal_plan','meals') is not none %} {{prefix}}{{url}}{{suffix}} {% endif %}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.dinner_today_image'
TemplateError('UndefinedError: None has no element 0') while processing template 'Template<template=({% set prefix='https://edwardshomeassist.duckdns.org/api/hassio_ingress/3-NJxerzOzNSyTRa3APRHLRxNW9T23LPMa095OjeME4' %} {% set url=state_attr('sensor.grocy_meal_plan','meals')[0].picture_url | replace("/grocy/", "/files/",1) %} {% set suffix='?force_serve_as=picture&best_fit_width=400' %} {% if state_attr('sensor.grocy_meal_plan','meals') is not none %} {{prefix}}{{url}}{{suffix}} {% endif %}) renders=1>' for attribute '_attr_entity_picture' in entity 'sensor.dinner_today_image'`

and (i think this one is because the source value is not present)

`Logger: homeassistant.helpers.event
Source: helpers/template.py:646
First occurred: 5:06:31 PM (2 occurrences)
Last logged: 5:06:31 PM

Error while processing template: Template<template=({{ (states('sensor.inverter_inverter') | replace(",", "") | float) - (states('sensor.pv_excess') | replace(",", "") | float) }}) renders=2>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2443, in forgiving_float_filter
    return float(value)
ValueError: could not convert string to float: 'unknown'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 644, in async_render
    render_result = _render_with_context(self.template, compiled, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2969, in _render_with_context
    return template.render(**kwargs)
           ~~~~~~~~~~~~~~~^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/jinja2/environment.py", line 1295, in render
    self.environment.handle_exception()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/jinja2/environment.py", line 942, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2446, in forgiving_float_filter
    raise_no_default("float", value)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1972, in raise_no_default
    raise ValueError(
    ...<2 lines>...
    )
ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.inverter_inverter') | replace(",", "") | float) 
- (states('sensor.pv_excess') | replace(",", "") | float) }}' but no default was specified

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 761, in async_render_to_info
    render_info._result = self.async_render(  # noqa: SLF001
                          ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
        variables, strict=strict, log_fn=log_fn, **kwargs
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 646, in async_render
    raise TemplateError(err) from err
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.inverter_inverter') | replace(",", "") | float) 
- (states('sensor.pv_excess') | replace(",", "") | float) }}' but no default was specified`

and

`Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:657
integration: Sensor (documentation, issues)
First occurred: 5:06:28 PM (1 occurrence)
[... 57 lines cut ...]
Last logged: 5:06:28 PM

Error while setting up sems platform for switch: 'name'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 382, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/config/custom_components/sems/switch.py", line 26, in async_setup_entry
    async_add_entities(
    ~~~~~~~~~~~~~~~~~~^
[... cut ...]
    new_entities if type(new_entities) is list else list(new_entities)
                                                    ~~~~^^^^^^^^^^^^^^
  File "/config/custom_components/sems/switch.py", line 27, in <genexpr>
    SemsStatusSwitch(coordinator, ent) for idx, ent in enumerate(coordinator.data)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/sems/switch.py", line 61, in __init__
    name=f"Inverter {self.coordinator.data[self.sn]['name']}",
                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
KeyError: 'name'`

Thanks

TimSoethout, 2025-06-22:
The only change was adding these debug statements:
```
        _LOGGER.debug("state, coordinator data: %s", self.coordinator.data)
        _LOGGER.debug("self.sn: %s", self.sn)
        _LOGGER.debug("state, self data: %s", self.coordinator.data[self.sn])
        data = self.coordinator.data[self.sn]
        return data["etotal"]
```
So no expected change, but hopefully more information. The debug statements however I don't see your posts. Maybe one of you can enable debug logging and posting the relevant part of the log? 
The HA UI by default only shows the error it seems.

justinmaiuto, 2025-06-23:
Just tried 7.4.0 my phone so hard to get logs, but still broken it seems. Current house consumption was tge main entity I used previously. The Homekit xxx entity is showing a watt value, but it is not changing/updating

[image]

justinmaiuto, 2025-06-23:
Logs i could spot:

Logger: homeassistant.components.template.template_entity
Source: components/template/template_entity.py:215
integration: Template (documentation, issues)
First occurred: 06:03:38 (2 occurrences)
Last logged: 06:03:38

TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ states ('sensor.inverter_5010kmst22bw0458') | float | round(0) - states ('sensor.homekit_91000hku22a50161') | float | round(0)}}' but no default was specified') while processing template 'Template<template=({{ states ('sensor.inverter_5010kmst22bw0458') | float | round(0) - states ('sensor.homekit_91000hku22a50161') | float | round(0)}}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.current_grid_export_import'
TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{% if states('sensor.current_grid_export_import') | float | round(0) >= 0 %} {{ states('sensor.inverter_5010kmst22bw0458') | float | round(0) - states('sensor.current_grid_export_import') | float | round(0) }} {% elif states('sensor.current_grid_export_import') | float | round(0) < 0 %} {{ states('sensor.homekit_91000hku22a50161') | float | round(0) + states('sensor.current_grid_export_import') | float | round(0) }} {% endif %}' but no default was specified') while processing template 'Template<template=({% if states('sensor.current_grid_export_import') | float | round(0) >= 0 %} {{ states('sensor.inverter_5010kmst22bw0458') | float | round(0) - states('sensor.current_grid_export_import') | float | round(0) }} {% elif states('sensor.current_grid_export_import') | float | round(0) < 0 %} {{ states('sensor.homekit_91000hku22a50161') | float | round(0) + states('sensor.current_grid_export_import') | float | round(0) }} {% endif %}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.current_solar_self_consumption'

Logger: homeassistant.helpers.event
Source: helpers/template.py:646
First occurred: 06:03:38 (2 occurrences)
Last logged: 06:03:38

Error while processing template: Template<template=({{ states ('sensor.inverter_5010kmst22bw0458') | float | round(0) - states ('sensor.homekit_91000hku22a50161') | float | round(0)}}) renders=2>
Error while processing template: Template<template=({% if states('sensor.current_grid_export_import') | float | round(0) >= 0 %} {{ states('sensor.inverter_5010kmst22bw0458') | float | round(0) - states('sensor.current_grid_export_import') | float | round(0) }} {% elif states('sensor.current_grid_export_import') | float | round(0) < 0 %} {{ states('sensor.homekit_91000hku22a50161') | float | round(0) + states('sensor.current_grid_export_import') | float | round(0) }} {% endif %}) renders=2>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2443, in forgiving_float_filter
    return float(value)
ValueError: could not convert string to float: 'unavailable'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 644, in async_render
    render_result = _render_with_context(self.template, compiled, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2969, in _render_with_context
    return template.render(**kwargs)
           ~~~~~~~~~~~~~~~^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/jinja2/environment.py", line 1295, in render
    self.environment.handle_exception()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/jinja2/environment.py", line 942, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2446, in forgiving_float_filter
    raise_no_default("float", value)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1972, in raise_no_default
    raise ValueError(
    ...<2 lines>...
    )
ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ states ('sensor.inverter_5010kmst22bw0458') | float | round(0)
 - states ('sensor.homekit_91000hku22a50161') | float | round(0)}}' but no default was specified

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 761, in async_render_to_info
    render_info._result = self.async_render(  # noqa: SLF001
                          ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
        variables, strict=strict, log_fn=log_fn, **kwargs
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 646, in async_render
    raise TemplateError(err) from err
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ states ('sensor.inverter_5010kmst22bw0458') | float | round(0)
 - states ('sensor.homekit_91000hku22a50161') | float | round(0)}}' but no default was specified

This error originated from a custom integration.

Logger: custom_components.sems.sensor
Source: custom_components/sems/sensor.py:185
integration: GoodWe SEMS API (documentation, issues)
First occurred: 06:03:26 (1 occurrence)
Last logged: 06:03:26

Skip migration of id [5010KMST22BW0458] to [5010KMST22BW0458-power] because it already exists

hewball, 2025-06-24:
here is a debug reload of working version with homekit 

[2025-06-24 182413.798 DEBUG.txt](https://github.com/user-attachments/files/20881754/2025-06-24.182413.798.DEBUG.txt)

daveedwards123, 2025-06-25:
With 7.4.0-beta it's working now, however the sensors were renamed:
- Grid Import -> Import
- Solar Export -> Export
- Energy removed
I just had to update a new of my related templates and Energy Dashboard

Many thanks @TimSoethout

justinmaiuto, 2025-06-25:
Ok, looks like it was an issue with polling for me, I had disabled polling and did it manually to workaround an issue with the SEMS side where there is some double counting not long after midnight on their side. The polling automation I setup referenced the old house consumption entity. I'll double check it all works now.

justinmaiuto, 2025-06-25:
Confirmed, working all good!

hewball, 2025-06-25:
7.4.0-beta works for me too, Energy sensor changed, but working as expected

TimSoethout, 2025-06-25:
Cool, thanks for checking everyone!
Before I release this beta as 4.7.0 then, should I fix some naming for?
Grid Import -> Import
Solar Export -> Export

davidsonimagerygmailcom, 2025-06-27:
Ahh this thread explains why mine broke :) ... all good reverted HA to a backup and will stay on 6.1.1 until 7.4.0 hits the floor :)

AgentKith, 2025-06-30:
> 7.4.0-beta works for me too, Energy sensor changed, but working as expected

Hi, do you mind showing the sensors for your homekit? I tried 7.4.0-beta and I am unsure what the old engery sensor has changed to. Also, are you using automatic HA polling, or are you using an automation to get the data (to work around the midnight spike?).

I'm using the homekit to find out the total power consumption in my house, and for some reason I can't seem to find that sensor somehow (very likely user error).

thanks.

In the meantime I have reverted back to 6.11

chicaneau, 2025-06-30:
I lost my 'house consumption' sensor overnight.
Then today i've upgraded to 7.40beta to see if it would fix it. However it hasnt appeared again. I even removed the integration and setup from scratch but it didnt help.

I did notice that my regular SEMS portal iOS app also isn't showing the house consumption graph. But the SEM+ and online portal are showing it. Perhaps SEMS has deprecated something in migrating to SEMS+?

AgentKith, 2025-06-30:
> I lost my 'house consumption' sensor overnight. Then today i've upgraded to 7.40beta to see if it would fix it. However it hasnt appeared again. I even removed the integration and setup from scratch [...]

I've reverted back to 6.11 and the home consumption sensor is working for me, and the data more or less matches SEMS+.

With 6.11, I get 3 sensors (SN is the serial number of my homekit):
* SN (Home consumption)
* SN Import (Energy imported from the grid)
* SN Export (Energy exported to the grid)

Will try a new version of this plugin over the weekend. IIRC only the export sensor was working for me. Import might not be working as  the solar system hasn't kicked in yet.. And I expect energy to be the consumption but that's not working. Yet to check the logs to see what's wrong.

TimSoethout, 2025-07-01:
> When is 7.4 scheduled for release? Deciding whether to downgrade or wait

I will try to release it this morning. There were some merge issues.

TimSoethout, 2025-07-01:
Released:

https://github.com/TimSoethout/goodwe-sems-home-assistant/releases/tag/7.4.0

Let's close this for now, reopen a new issue for specifics if needed. :)