TimSoethout/goodwe-sems-home-assistant #138: Unable to fetch data from SEMS.

LucasSos, 2025-09-08:
Hello,

I am using the GoodWe SEMS API, version 7.4.0 with GoodWe GW8KN-ET.
During configuration I get the following error:
Failure; retrying configuration: Error communicating with API: 'inverter'.

How can I solve this?

MajoV, 2025-09-08:
I have the same problem.

Chrisjacobs62, 2025-09-08:
I'am getting also the same error since yesterday.
In the logfile repeatly appears: [custom_components.sems.sems_api] Unable to fetch data from SEMS and debug log shows:
2025-09-08 10:45:16.236 DEBUG (ImportExecutor_0) [custom_components.sems.config_flow] SEMS - new config flow
2025-09-08 10:45:16.238 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] SEMS - Making Power Station Status API Call
2025-09-08 10:45:16.239 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] API token not set (None) or new token requested (False), fetching
2025-09-08 10:45:16.239 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] SEMS - Getting API token
2025-09-08 10:45:16.527 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] Login Response: <Response [200]>
2025-09-08 10:45:16.527 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] SEMS - API Token received: {'uid': 'xxxx-xxxx-xxxx-xxxx-xxxx', 'timestamp': 1757321116442, 'token': 'yyyyyyyyyy', 'client': 'ios', 'version': '', 'language': 'en', 'api': 'https://eu.semsportal.com/api/'}
2025-09-08 10:45:16.528 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] Querying SEMS API (https://eu.semsportal.com/api//v3/PowerStation/GetMonitorDetailByPowerstationId) for power station id: xxx-xxxx-xxxx-xxxx-xxxx
2025-09-08 10:45:17.507 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] Response: {'language': 'en', 'function': ['ADD', 'VIEW', 'EDIT', 'DELETE', 'INVERTER_A', 'INVERTER_E', 'INVERTER_D'], 'hasError': False, 'msg': '操作成功', 'code': '0', 'data': {'info': {'powerstation_id': 'xxx-xxxx-xxxx-xxxx-xxxx', 'time': '09/08/2025 10:45:16', 'date_format': 'dd.MM.yyyy', 'date_format_ym': 'MM.yyyy', 'stationname': 'zzzzz, 'address': 'zzzzz, zzzzz', 'owner_name': 'zzzzzzl', 'owner_phone': None, 'owner_email': 'zzzzz', 'battery_capacity': 0.0, 'turnon_time': '11/24/2022 00:00:00', 'create_time': '11/24/2022 23:10:39', 'capacity': 5.0, 'longitude': zzzzz, 'latitude': zzzz, 'powerstation_type': 'Residential', 'status': 1, 'is_stored': False, 'is_powerflow': False, 'charts_type': 4, 'has_pv': True, 'has_statistics_charts': False, 'only_bps': False, 'only_bpu': False, 'time_span': -2.0, 'pr_value': '', 'org_code':  
etc.etc.(response seems to be correct)
2025-09-08 10:45:17.509 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] Query not successful (操作成功), retrying with new token, 2 retries remaining

snhnic, 2025-09-08:
Yep same here. Tried to update with no results

krek8, 2025-09-08:
same her, since last night 4.14

glaubernobrega, 2025-09-08:
I have the same problem.

SMac3265, 2025-09-08:
It looks like the success message changed, perhaps? I made this patch to custom_components/sems/sems_api.py and I'm getting data flowing again:

OLD CODE (starts at line 117):

            if jsonResponse["msg"] != "success" or jsonResponse["data"] is None:
                _LOGGER.debug(
                    "Query not successful (%s), retrying with new token, %s retries remaining",
                    jsonResponse["msg"],
                    maxTokenRetries,
                )
                return self.getData(
                    powerStationId, True, maxTokenRetries=maxTokenRetries - 1
                )

            return jsonResponse["data"]

NEW CODE:

            # Parse JSON safely
            try:
                jsonResponse = response.json()
            except ValueError as e:
                _LOGGER.debug("SEMS API returned non-JSON: %s", e)
                return self.getData(
                    powerStationId,
                    True,
                    maxTokenRetries=maxTokenRetries - 1,
                )

            msg = str(jsonResponse.get("msg", "")).strip()
            msg_lc = msg.lower()
            code = jsonResponse.get("code", None)
            data = jsonResponse.get("data", None)

            # Consider success if numeric code==0 OR msg is a known success string (any locale)
            ok_code = code in (0, "0")
            ok_msg = (
                msg_lc in ("success", "successful")
                or msg in ("操作成功",)
            )

            if not (ok_code or ok_msg) or data is None:
                _LOGGER.debug(
                    "SEMS API unexpected response: code=%s msg=%r data_is_none=%s; retrying, %s retries left",
                    code,
                    msg,
                    data is None,
                    maxTokenRetries,
                )
                return self.getData(
                    powerStationId,
                    True,
                    maxTokenRetries=maxTokenRetries - 1,
                )

            return data

BozzaJM, 2025-09-09:
Thanks for the quick fix!!

sagwefwefga, 2025-09-09:
Fixed for me, I not sure what happened but the name of my entity changed so went into yaml and changed the entity name of all my sensors and it’s working again